robotpy/pynetworktables #21: NT 3.0 support

PeterJohnson, 2015-12-01:
Need to update this for NT 3.0 compatibility.  WPILib Java/C++ was basically a complete rewrite, so this could be done as either a translation of that rewrite into Python or by updating this code with the NT 3.0 protocol.

virtuald, 2015-12-01:
My preference is to keep pynetworktables as pure python, to make it easier for people to use this on the various platforms that they use (windows, linux, osx, beaglebones, etc). Of course, that means a lot more work...

I suspect a translation of the rewrite would be easier.

virtuald, 2015-12-01:
I would like to call attention to the [lock wrappers](https://github.com/robotpy/pynetworktables/blob/master/networktables2/_impl_debug.py) created to debug the current implementation -- they're really useful for testing for deadlocks, so any rewrite should try to use those I think (refactoring the lock ordering, of course).

ryannazaretian, 2016-02-03:
Any updates to this?

virtuald, 2016-02-03:
@PeterJohnson said he was going to do a compatibility migration this week.

virtuald, 2016-02-24:
For those who want to use this, you can install the alpha version of pynetworktables:

    pip install -U --pre pynetworktables

ryannazaretian, 2016-03-04:
Finally got around to trying it, but not having much luck. 

First issue was with leb128.py trying to shift a string with an int. If you type(rstream.read(1)[0]), it'll return string. I tried to fix it by wrapping rstream.read(1)[0] with int(), but apparently rstream.read(1)[0] can contain stuff other than numbers because I received another error about rstream.read(1)[0] not being a base10 value. That only happened once though and I was unable to determine what the data actually was. 

After doing the wrapper int(rstream.read(1)[0]), it ran as if it was working; however, the variables were never written to the NT table server.

virtuald, 2016-03-05:
What kind of type are you trying to send/receive?

virtuald, 2016-03-05:
Also, keep in mind that this can only talk to an NT 3.0 server, so if you try to talk to a 2.0 server it will fail.

ryannazaretian, 2016-03-08:
Using LabVIEW, so NT 3.0. I tried sending various types of data including boolean, numeric, string, boolean array, numeric array, and string array.

virtuald, 2016-03-08:
Which version of python are you using, and which particular types are you having problems with (all of them, some of them?). Someone else had success with this.

ryannazaretian, 2016-03-08:
I think I tried both 2.7 and 3. One wouldn't import if I remember correctly. I'm having problems with all types.

virtuald, 2016-03-08:
One last question -- are you sure that it's successfully connecting to the server? If you enable python logging (see the samples), there's a log message that will get emitted when the connection is successful. If that doesn't happen, then you should address that problem first.

Do you have some simple sample code that doesn't work for you (or have you tried the provided samples?).

ryannazaretian, 2016-03-08:
I'll have to give it another try. My memory isn't that great on what I was doing and what the code wasn't doing. It might be sometime next week before I can try again.

virtuald, 2016-03-08:
Not a problem, I'll have more time to look at it next week anyways.

ryannazaretian, 2016-03-26:
Obviously stress and exhaustion got to me...

I didn't install using pip3. I have other Python code that needs to run under 2.7. Giving stuff a try now. Fingers crossed.

virtuald, 2016-04-05:
@ryannazaretian see #26, looks like it's a python 2.7 specific problem.

virtuald, 2016-04-05:
Also, I fixed it. Install 2016.0.0a3 and it should work for you. Sorry for the delay.

virtuald, 2016-09-18:
Initial ntcore porting work is now happening in the nt3 branch. Not anywhere near comprehendable yet, but it's amazing how far you can get using a regex-based converter (cpp2python).

virtuald, 2016-10-25:
The rewrite is 90% done at this point, and works under python 2 and python 3, and can connect to NT2 and NT3 clients. Once I fix ntproperty,  the global table stuff, and some misc fixes, I'll push a beta release for people to play with... probably tomorrow night.

virtuald, 2016-10-28:
Fixed in fa687a08d2e957db72b5a959d907bccd1648e116, please try it out and let me know if it works for you, thanks!

closed 2016-10-28 by virtuald