robotpy/pynetworktables #22: Python Network Tables with GRIP

codetheweb, 2016-01-20:
Our team is trying out [GRIP](https://github.com/WPIRoboticsProjects/GRIP), but I can't seem to get the data that is published over network tables by it.  I have successfully printed the data in C++ with something like `double x = grip->GetNumberArray("ball/x", llvm::ArrayRef<double>())` (where `grip` is the network table and `ball` is a subtable).  However, I have tried doing the same in Python with various failure error messages:

`getData()`: no longer seems to exist, but still in docs
`getDouble()`: returns `TypeError: Cannot get number for '/GRIP/ball/x', is a Array of [Double]`
`getNumber()`: returns exact same thing as `getDouble()`.

Is there anything else I should try?  I confirmed that the data is in network tables with the TableViewer utility.

virtuald, 2016-01-20:
Try out [retrieveValue](http://pynetworktables.readthedocs.org/en/latest/networktables.html#networktables.networktable.NetworkTable.retrieveValue), should do the trick.

virtuald, 2016-01-20:
Also, which docs is getData still in? I'd like to remove it if it's no longer there.

codetheweb, 2016-01-20:
[Here](http://robotpy.readthedocs.org/en/latest/wpilib/SmartDashboard.html#module-wpilib.smartdashboard)

codetheweb, 2016-01-23:
Still doesn't seem to be working.  Here's my code:
```
import networktables
from networktables import NetworkTable

ball = NetworkTable.getTable('/GRIP/ball')
valType = networktables.NumberArray()
ballX = ball.retrieveValue("x", valType)
```

`ballX` always has type `None`, although I can confirm with the TableViewer that `GRIP/ball/x` has a value.

virtuald, 2016-01-23:
They are using NT 3.0, so perhaps the implementation has changed and it's not backwards compatible to 2.0. 

Try using the [global listener](https://github.com/robotpy/pynetworktables/blob/master/samples/global_listener.py) to see what values show up. If it doesn't show up, then that certainly could be the problem. If it does show up, then we can debug it to figure out why you can't get at it.

codetheweb, 2016-01-23:
Yay, finally works!  Spits out stuff like this:
```
valueChanged: key: '/GRIP/ball/size'; value: [1.6180340051651]; isNew: False
valueChanged: key: '/GRIP/ball/x'; value: [49.5]; isNew: False
valueChanged: key: '/GRIP/ball/y'; value: [137.0]; isNew: False
```

So how can I access that from my program?

PeterJohnson, 2016-01-23:
That's not how retrieveValue() works.  It puts the result in the passed object, and does not return anything.  You want something like:
```python
ball = NetworkTable.getTable('/GRIP/ball')
ballX = networktables.NumberArray()
ball.retrieveValue("x", ballX)
```
You might also need to remove the leading "/" from the table name, so "GRIP/ball" instead of "/GRIP/ball".

codetheweb, 2016-01-23:
It works!  Thanks @PeterJohnson and @virtuald.

closed 2016-01-23 by codetheweb