robotpy/pynetworktables #26: Connection Problem with Python 2.7 and 2016.0.0a2 Release

PChild, 2016-04-01:
Started testing possibly using on our NVIDIA Jetson and am getting this error. Testing with other machines also throws this error with Python 2.7, but everything works with Python 3.5. Any ideas how I can fix this? I can add any additional information that may be useful.

```
root@tegra-ubuntu:/home/ubuntu/Desktop/Code/2016-Robot-Code/Vision/main# python networkTest.py
INFO:nt:Client 0xb601e918 entered connection state: CONNECTED_TO_SERVER
Exception in thread Client Connection Reader Thread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/networktables2/connection.py", line 194, in run
    self.connection.read(self.adapter)
  File "/usr/local/lib/python2.7/dist-packages/networktables2/connection.py", line 102, in read
    serverName, (flags,) = SERVER_HELLO.read(self.rstream)
  File "/usr/local/lib/python2.7/dist-packages/networktables2/messages.py", line 71, in read
    nameLen = leb128.read_uleb128(rstream)
  File "/usr/local/lib/python2.7/dist-packages/networktables2/leb128.py", line 18, in read_uleb128
    result |= (b & 0x7f) << shift
TypeError: unsupported operand type(s) for &: 'str' and 'int'
```

virtuald, 2016-04-02:
Do you require LabVIEW support? If not, do:

```
pip install pynetworktables==2015.3.2
```

Probably you have an old version of pip, and it installed the alpha version by accident.

PChild, 2016-04-04:
Hi, yes, we are using Labview. I intentionally installed the alpha per discussion on Chief Delphi.

virtuald, 2016-04-05:
Sat down and finally fixed this, it (of course) was a difference in the way that python2 and 3 handled bytes... so I wrapped it in a bytearray and they're both happy. Added some unit tests for good measure, and tested python 2 and 3 against SmartDashboard.

Released as 2016.0.0a3

commit 3bd5231e (2016-04-05): Fix leb128 encoding, add tests

- Fixes #26
closed 2016-04-05 by virtuald