robotpy/pynetworktables #34: Install not working - missing files

drakoswraith, 2016-12-29:
It appears that installation isn't working correctly due to missing files in the archive

Using the "pip install networktables" method resulted in:
- the neworktables.py file is missing from the neworktables package folder
- the ntcore module is missing

I tested on a windows system and a raspberry pi with the same results.

[networktables install output.txt](https://github.com/robotpy/pynetworktables/files/677745/networktables.install.output.txt)

virtuald, 2016-12-30:
I suppose this should be documented, but the ntcore-based version of pynetworktables has not been officially released yet (will be released on kickoff morning). If you wish to use that version, you need to do `pip install --pre pynetworktables`.

The 'stable' documentation (which is default if you go to http://pynetworktables.readthedocs.io/) should document the version that is installed without the `--pre` flag. The 'latest' documentation shows the new API (http://robotpy.readthedocs.io/projects/pynetworktables/en/latest/).

The main reason for not releasing this version is because there are some incompatibilities with the current version of RobotPy, and I don't want to break that before a new release.

closed 2016-12-30 by virtuald

drakoswraith, 2016-12-30:
Thanks, i will try the other steps.
However, note that following the steps for "stable" release still resulted
in the networktables.py file missing, and prevented the import statements
in scripts from working. So i think there is still a potential issue
(hopefully resolved with the ntcore release as well)

Michael

virtuald, 2016-12-30:
The 'stable' release does not have a networktables.py file, that is a new change for 2017. If you're using the stable release, try out the scripts at http://pynetworktables.readthedocs.io/en/stable/examples.html