robotpy/pynetworktables #36: Problem running pynetworktables in python 2.7 with unicode characters

juanct, 2017-01-11:
We were using a raspberrypi running network tables for communication with the roborio, but when the roborio creates a table for microsoft camera which has a special character in its description the function NetworkTables.initialize(server=ip) cannot start connection because a UnicodeEncodeError during the handshake.

The debuging return the following errors:

DEBUG:nt:client connected
DEBUG:nt:NetworkConnection stopping (<ntcore.network_connection.NetworkConnection object at 0x74469330>)
ERROR:nt:Unhandled exception during handshake
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ntcore/network_connection.py", line 240, in _readThreadMain
    handshake_success = self.m_handshake(self, _getMessage, self._sendMessages)
  File "/usr/local/lib/python2.7/dist-packages/ntcore/dispatcher.py", line 488, in _clientHandshake
    msg = get_msg()
  File "/usr/local/lib/python2.7/dist-packages/ntcore/network_connection.py", line 228, in _getMessage
    return Message.read(self.m_stream, decoder, self.m_get_entry_type)
  File "/usr/local/lib/python2.7/dist-packages/ntcore/message.py", line 123, in read
    value = codec.read_value(value_type, rstream)
  File "/usr/local/lib/python2.7/dist-packages/ntcore/wire.py", line 114, in read_value
    return Value.makeString(self.read_string(rstream))
  File "/usr/local/lib/python2.7/dist-packages/ntcore/value.py", line 51, in makeString
    return ValueType(NT_STRING, str(value))
UnicodeEncodeError: 'ascii' codec can't encode character u'\xae' in position 9: ordinal not in range(128)
INFO:nt:DISCONNECTED roboRIO-3478-FRC.local port 1735 (Robot)
DEBUG:nt:write thread died (<ntcore.network_connection.NetworkConnection object at 0x74ce7230>)

[image]

virtuald, 2017-01-11:
Thanks for reporting the bug! This should be fairly straightforward to fix, I'll do it tonight.

virtuald, 2017-01-11:
Also, if you're impatient, I'd bet that Python 3 would handle this without any issues.

virtuald, 2017-01-12:
Fixed in 2017.0.2, thanks @auscompgeek for the fix

PR #37 (2017-01-12): ntcore.value: Use unicode strings in Python 2
Fixes #36.
closed 2017-01-12 by virtuald