robotpy/pynetworktables #40: Does not print "Disconnected" when robot disconnects

calebeby, 2017-01-27:
Hi Dustin,

First of all, thanks for this! I have found it really useful.

Here's the issue:

The robot connects, prints `CONNECTED`, `pynetworktables2js` sends value :white_check_mark:
The robot/network disconnects. Nothing happens :x:

Any idea why?

Thanks

calebeby, 2017-01-27:
The above happens on Ubuntu, but the behavior is as expected on Windows.

virtuald, 2017-01-29:
Interesting. So you're saying that pynetworktables2js doesn't send the event to the HTML page, and that nothing is showing up?

I'm curious if you see this issue just using the networktables examples.

calebeby, 2017-01-29:
Correct. I'll check the examples when I get access to robotics computers
tomorrow.

virtuald, 2017-04-15:
Assume this resolved itself? Closing.

closed 2017-04-15 by virtuald