robotpy/pynetworktables #49: Sync code with 2018 updates

virtuald, 2017-09-17:
This may be fairly involved, there are some updates to the way ntcore does synchronization. Additionally, they changed NT to have instances instead of lots of static globals -- but we do that anyways. Still, the code will need to be reviewed and updated.

virtuald, 2017-09-27:
I'm going to start on this tonight and see how bad the updates are, as this is blocking pull requests over in robotpy-wpilib.

auscompgeek, 2017-09-27:
Fixing #35 would probably be useful as well. Heck, there's a few other 2017 things that never landed here.

virtuald, 2017-09-27:
Well, done for tonight. I updated the git-source-track information, and *most* of the changes seem to be pretty minimal -- but there are major changes in storage and notifier (which has been split in 3), so it'll be quite a bit of work.

PR #51 (2017-10-23): Implement ntcore 2018 changes
Here's the initial updates. I haven't really done anything to the user facing APIs yet... so that remains to be done. Unfortunately, this took about a week longer than I had expected. I would appreciate any initial reviews/testing of this, there's quite a few changes -- but the tests finally pass!

- Fixes #49
closed 2017-10-23 by virtuald