robotpy/pynetworktables #54: Drop Python 3.3 support in 2019

auscompgeek, 2018-01-01:
I would like to propose we drop support for Python 3.3 in 2019.

Python 3.3 is already EOL (as of September 2017), and is not distributed by major, currently supported GNU/Linux distros, so I don't anticipate this being a problem for any team.

We've already announced on CD that pynetworktables will continue to support Python 3.3 for this year, so I guess we're keeping it this way.  I wouldn't imagine we'd have many more changes this year, so this probably won't be much of a problem for us anyway.

virtuald, 2018-01-01:
People use pynetworktables on weird embedded devices that may not have been updated in forever. I don't think we should drop support until there's a good reason to -- and it'll probably be dropped around the same time as Python 2.7 support...

auscompgeek, 2018-01-01:
It'd be good to see the PyPI download stats, I guess.

virtuald, 2018-05-18:
FWIW, it seems that 12 people on CD would like to drop Python 2 support. https://www.chiefdelphi.com/forums/showthread.php?p=1766160

auscompgeek, 2018-09-10:
From what I can tell, upstream ntcore is being rearchitected to use an event loop.

We've also been talking about potentially internally using asyncio instead of threads, hopefully improving performance in the process. If we were to go down that path, we'd likely want to be able to use async/await syntax, which would mean dropping support for all Python versions before 3.5.

If anyone cares about Python 3.4 support (or if you _really_ care about Python 3.3), please speak up now.

virtuald, 2018-11-23:
We will no longer support python < 3.5 in 2019.

commit 2d7a569a (2018-11-23): Add python 3.7 to travis, remove < 3.5

- Fixes #54
commit 66b2ead1 (2018-11-23): Add python 3.7 to travis, remove < 3.5

- Fixes #54
commit ce6a1213 (2018-11-23): Add python 3.7 to travis, remove < 3.5

- Fixes #54
closed 2018-11-23 by virtuald