robotpy/pynetworktables #59: 'EntryNotification' has no attribute 'getEntry'

ErikBoesen, 2018-02-06:
I've been continuously getting these errors in the DS console. Nothing appears to be failing, but it makes it really hard to debug other code since they fill up the console after a crash.

```
 09:33:51:575 WARNING : nt                  : Unhandled exception processing entry-notifier callback 
 Traceback (most recent call last): 
   File "/usr/local/lib/python3.6/site-packages/ntcore/callback_manager.py", line 116, in main 
     self.doCallback(listener.callback, item) 
   File "/usr/local/lib/python3.6/site-packages/ntcore/entry_notifier.py", line 79, in doCallback 
     callback(data) 
   File "/usr/local/lib/python3.6/site-packages/cscore/cameraserver.py", line 367, in _onTableChange 
     event.getEntry().setNumber(prop.get()) 
 AttributeError: 'EntryNotification' object has no attribute 'getEntry'
```

virtuald, 2018-02-06:
This is a cscore bug, fixed in 2018.0.3 (will publish in a few minutes).

closed 2018-02-06 by virtuald

virtuald, 2018-02-06:
Releases published.

auscompgeek, 2018-02-07:
I am personally still of the opinion that entry events should have a getEntry. 🤷‍♂️

virtuald, 2018-02-07:
They're namedtuples, they can't.