robotpy/pynetworktables #61: pynetworktables initialize throws errors about utf-8 encoding

PeterMitrano, 2018-03-27:
I tried this:

```
logging.basicConfig(level=logging.DEBUG)
NetworkTables.initialize(server="10.6.66.5")
```

And got this:

```
INFO:nt:NetworkTables initialized in client mode
DEBUG:nt.th:Started thread nt-dispatch-thread-0
DEBUG:nt.th:Started thread nt-client-thread-0
DEBUG:nt:client connected
DEBUG:nt.th:Started thread nt-net-write-0
DEBUG:nt.th:Started thread nt-net-read-0
ERROR:nt:Unhandled exception during handshake
Traceback (most recent call last):
  File "C:\Users\peter\Anaconda3\lib\site-packages\ntcore\network_connection.py", line 248, in _readThreadMain
    handshake_success = self.m_handshake(self, _getMessage, self._sendMessages)
  File "C:\Users\peter\Anaconda3\lib\site-packages\ntcore\dispatcher.py", line 571, in _clientHandshake
    msg = get_msg()
  File "C:\Users\peter\Anaconda3\lib\site-packages\ntcore\network_connection.py", line 236, in _getMessage
    return Message.read(self.m_stream, decoder, self.m_get_entry_type)
  File "C:\Users\peter\Anaconda3\lib\site-packages\ntcore\message.py", line 123, in read
    value = codec.read_value(value_type, rstream)
  File "C:\Users\peter\Anaconda3\lib\site-packages\ntcore\wire.py", line 114, in read_value
    return Value.makeString(self.read_string(rstream))
  File "C:\Users\peter\Anaconda3\lib\site-packages\ntcore\wire.py", line 199, in read_string_v3
    return rstream.read(slen).decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xea in position 9: invalid continuation byte
INFO:nt:DISCONNECTED 10.6.66.5 port 1735 (Robot)
DEBUG:nt.th:Thread nt-net-read-0 exited
DEBUG:nt.th:Thread nt-net-write-0 exited
```

virtuald, 2018-03-27:
What version of python is it? And is it the latest version of pynetworktables?

virtuald, 2018-03-27:
Also, what are you connecting to? LabVIEW, Java, C++ ... and what version?

virtuald, 2018-03-27:
This looks very similar to https://github.com/robotpy/pynetworktables/issues/42, are you sure you're not using an old version?

PeterMitrano, 2018-03-27:
I'm using `pynetworktables==2018.1.0`

PeterMitrano, 2018-03-27:
I can't reproduce it locally just with outline viewer in server mode, so it's possible the network table running on the robot is somehow out of date. I know it has one of the valid releases of wpilib from this season

virtuald, 2018-03-27:
It's most likely a weird key that's being incorrectly encoded that's causing issues. Are you using camera server? Any chance you can connect with outline viewer and look at the camera name or description and see if there's something weird in it?

virtuald, 2018-03-27:
I wrote a stream debugger last year... never pushed it to master for some reason. I've updated it and pushed it to the `handshake-2018` branch. Please install that... if an exception is thrown, then it will write the stream to a file called 'file.bin'. Attach that here and I'll take a look tomorrow.

virtuald, 2018-03-28:
Let me know when I can expect to receive a `file.bin` from you, and I'll allocate some time in my schedule to take a look at it.

PeterMitrano, 2018-03-30:
I'm gonna try this out now--thanks for the quick response

virtuald, 2018-04-06:
Any news?

A similar bug has been reported on CD: https://www.chiefdelphi.com/forums/showthread.php?p=1754568

PeterMitrano, 2018-04-06:
No I realized I need the physical robot to test and haven't had a chance

Kython89, 2018-04-07:
In 2481's camera name there is a circle R (restricted) for the Microsoft live cam.  Would that cause problems?

auscompgeek, 2018-04-07:
That should be handled correctly, but it's possible it isn't.

(BTW, ® is the registered trademark symbol, not restricted.)

Kython89, 2018-04-07:
Okay.  That was the only thing that stood out as strange to me in the keys / values.

I already sent this Dustin but I'll include it here too.
[file.bin.zip](https://github.com/robotpy/pynetworktables/files/1885986/file.bin.zip)

virtuald, 2018-04-07:
Sorry for the delay, I was at a competition tonight and then stuff afterwards. I'm taking a look now, I'm able to reproduce the crash by sending the file through.

auscompgeek, 2018-04-07:
Update: from the file.bin that @Kython89 attached, we have concluded that the robot code is corrupting the value at key /LiveWindow/DriveTrain/.default from 'DriveWithJoystickCommandV2' to `b'\x00\xa7>\x03eWithJoystickCommandV2'`. Somehow. Not sure how that's happening, could be a WPILibC bug. We'll keep digging.

closed 2018-04-08 by virtuald

virtuald, 2018-04-08:
For those interested, somehow the memory of a string was being corrupted and passed into NetworkTables (this is still under investigation). As a result, the string received by pynetworktables was not valid utf-8 (as the NT spec dictates), which caused the issue.

PR #62 (2018-04-08): Gracefully handle invalid UTF-8 strings
From discussion on Gitter:

```irclog
<virtuald> I'm not really sure what pynetworktables should do in this
           case though
<virtuald> Peter thinks we should just gracefully handle it
<virtuald> any chance you can put together a PR that catches invalid
           string decodes and replaces the string with
           'INVALID UTF-8: <repr of bytes>'
<virtuald> ... that will probably crash his robot though if the string
           ever gets sent to the robot
<auscompgeek> yeah... stuck between a rock and a hard place here
<virtuald> so, java et al  [... cut]