robotpy/pynetworktables #84: issue running as server 'read error in handshake: end of file'

dbadb, 2019-02-01:
platform windows 10, python3.7.2 native build,  pynetworktables version 2019.0.0

in one git-bash window, i start a pynetworktables server:

`networktables.NetworkTables.initialize()`

in another git-bash window, I start a pynetworktables client:

 `networktables.NetworkTables.initialize(server="localhost")`

back in the server window, i see a stream like this:

```
read error in handshake: end of file
read error in handshake: end of file
read error in handshake: end of file
read error in handshake: end of file
read error in handshake: end of file
```

virtuald, 2019-02-01:
Hm. I don't know what's happening here. Did you enable logging as shown in the examples? Set the level to DEBUG and perhaps that output will be more useful.

```
# To see messages from networktables, you must setup logging
import logging
logging.basicConfig(level=logging.DEBUG)
```

Also, what is your client doing? If it exits immediately, then that's your problem. Try the [nt_driverstation.py](https://github.com/robotpy/pynetworktables/blob/master/samples/nt_driverstation.py) example.

virtuald, 2019-02-01:
I've no idea. Do you perhaps have a firewall enabled that might be killing the connection oddly? Unfortunately, I don't have a Windows 10 machine available to me at this time that I could test this with.

dbadb, 2019-02-01:
Hm - I had thought of that and had whitelisted python.  But just to be certain, i just disabled all three firewalls (public, private and domain) and still see the same behavior.   I'll keep you posted if I learn more.
thx!

dbadb, 2019-02-01:
btw: I can confirm that the same combination of scripts works on my mac laptop

virtuald, 2019-02-15:
Any news here?

dbadb, 2019-02-15:
no new news with all the excitement of the build season.
i did just re-verify that the condition persists on my windows 10 machine with python37 (windows native) plus pynetworktables  2019.0.0

virtuald, 2019-02-15:
I just setup a clean Windows 10 VM, installed Python 3.7.2, installed pynetworktables from pip, and ran the nt_robot.py / nt_driverstation.py programs. The firewall asked me if I wanted to allow the programs to communicate, and I said yes. They were able to send values back and forth without any problem.

I'm guessing you have something on your machine that is causing this issue. Recommend that you double-check your firewall configuration, and disable any antivirus.

dbadb, 2019-02-15:
completely agree that it smells machine specific.  as i mentioned earlier,
i believe i've disproved the theory that it's firewall related.
It would be interesting if the problem turns out to be related to the
national instruments installation.