robotpy/pynetworktables #93: pynetworktables installed, getting error import error

KenwoodFox, 2019-07-03:
Im getting this error
`ImportError: No module named 'networktables'`
when running any of the sample code

pip freeze shows
`pynetworktables==2019.0.2`
Thank you

auscompgeek, 2019-07-04:
What version of Python are you using (`python --version`), and which version of Python did you install it under (`pip --version`)?

KenwoodFox, 2019-07-04:
`$ python --version
Python 2.7.12
$ pip --version
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
$ python3 --version
Python 3.5.2
ubuntu@tegra-ubuntu:~/LAPIS$
`
Thanks so much for replying

andrewda, 2019-07-04:
pynetworktables is only compatible with Python 3.5 or later (see README). You'll have to install it with either `pip3 install pynetworktables` or `python3 -m pip install pynetworktables`, then run your code with `python3 some_networktables_program.py`

closed 2019-07-04 by auscompgeek

KenwoodFox, 2019-07-05:
Thank you so much! This fixed my problem.