robotpy/pynetworktables #99: problem running getDoubleArray

KhanSimeoni, 2020-01-30:
Hello,
I have been trying to use getDoubleArray, but I keep getting an error. 
I am using it in the following code:

```
table = NetworkTables.getTable("ML")

while 1:
    object1posX = table.getDoubleArray('boxes',0)
```
Unfortunatly, this returns the error: 
```
AttributeError: 'NetworkTable' object has no attribute 'getDoubleArray'
```
Any idea what i'm doing wrong?

virtuald, 2020-01-30:
Use `getNumberArray` instead.

closed 2020-01-30 by virtuald

KhanSimeoni, 2020-01-30:
That returned the error
```
TypeError: 'int' object is not subscriptable
```

virtuald, 2020-01-30:
What's the full stack trace?

KhanSimeoni, 2020-01-30:
Is this what your looking for?
```
python3 test_network_receive.py 10.17.21.2
INFO:nt:NetworkTables initialized in client mode
DEBUG:nt.th:Started thread nt-dispatch-thread-0
DEBUG:nt.th:Started thread nt-client-thread-0
Traceback (most recent call last):
  File "test_network_receive.py", line 23, in <module>
    print("The center is at " + str((object1posX[3]-object1posX[1])+object1posX))
TypeError: 'int' object is not subscriptable
```

virtuald, 2020-01-30:
The default value that you passed to `getNumberArray` is an integer, which is returned if the value isn't present in NetworkTables.

Try `getNumberArray('foo', [0,0,0,0])` instead. Or use `None` as the default, and check to see if the value is `None` and don't print if it is.

KhanSimeoni, 2020-01-30:
Got an `NameError: name 'getNumberArray' is not defined` error
```

INFO:nt:NetworkTables initialized in client mode
DEBUG:nt.th:Started thread nt-dispatch-thread-0
DEBUG:nt.th:Started thread nt-client-thread-0
Traceback (most recent call last):
  File "test_network_receive.py", line 22, in <module>
    object1posX = getNumberArray('foo', [0,0,0,0])
NameError: name 'getNumberArray' is not defined

```

virtuald, 2020-01-31:
... `table.getNumberArray`