robotpy/pynetworktables #105: Update rate does not match pyntcore

auscompgeek, 2020-02-07:
If pyntcore is installed, the default update rate will suddenly change, which I would expect nobody installing random packages would expect to happen.

The following line needs to be changed to be every 100 ms instead:

https://github.com/robotpy/pynetworktables/blob/master/_pynetworktables/_impl/dispatcher.py#L71

auscompgeek, 2020-02-07:
In the meantime, if you would like your update rate to be consistent regardless of what packages are installed, you can use the [`NetworkTablesInstamce.setUpdateRate`](https://robotpy.readthedocs.io/projects/pynetworktables/en/stable/api.html#networktables.NetworkTablesInstance.setUpdateRate) method.

Robot code always uses pyntcore, and so will always have a default update rate of 100 ms.

virtuald, 2020-02-07:
I feel like we should change pyntcore's default to be 50ms?

auscompgeek, 2020-02-07:
Given the performance issues we had last year, I'm not sure I'm comfortable with doubling down on the doubled update rate here.

auscompgeek, 2021-12-28:
Well, it's almost 2022. Which way do we want to go? I'm still in favour of matching ntcore here.

virtuald, 2021-12-28:
FIne.