robotpy/pynetworktables #112: Tests fail using ntcore backend

TheTripleV, 2020-07-21:
The following tests failed repeatedly on the ntcore backend:

- test_util.py::test_ntproperty[True] - assert None == 0
- test_util.py::test_ntproperty[False] - assert None == 0
- test_util.py::test_ntproperty_multitest[True] - assert None == 0
- test_util.py::test_ntproperty_multitest[False] - assert None == 0
- test_util.py::test_chooser_control[True] - AssertionError: assert None...
- test_util.py::test_chooser_control[False] - AssertionError: assert 'fo...

auscompgeek, 2020-07-23:
Took a look at this. Looks like we need to change line 20 of tests/conftest.py back to import from networktables, otherwise the utils will use a different backend to the test fixtures.

auscompgeek, 2020-07-23:
This might also be an indicator that we should actually pass the NetworkTablesInstance to the util classes in the tests as well.

auscompgeek, 2021-02-03:
Fixed by #115 and #118, but perhaps we should also make it possible to run the tests with the ntcore backend here?

TheTripleV, 2021-02-05:
This does run tests with ntcore.
The only issue is with some tests sporadically failing: https://github.com/robotpy/pynetworktables/actions/runs/498147113.
With the current matrix of size 32, jobs get queued making runs long.

auscompgeek, 2021-12-28:
> This does run tests with ntcore.

I believe early 2021 me was referring to the fact that none of the tests here will actually run against the ntcore backend, as the fixtures unconditionally use the pure-Python backend. Whilst I might agree that it's good to test that the pure-Python stuff still works with pyntcore installed, I'm not sure it's a good use of our CI time.

TheTripleV, 2021-12-28:
The reason for adding this testing was because team code written in pure python started failing when swapped to the pyntcore back end. I guess with nt4 coming in 2023, it doesn't really matter all too much anymore.