robotpy/pynetworktables #114: error 61 - connection refused

alpceylan, 2020-10-07:
I'm having this error and I can't connect to 127.0.0.1 because of this

`DEBUG:nt:client trying to connect`
`ERROR:nt:select() to 127.0.0.1 port 1735 error 61 - Connection refused`

virtuald, 2020-10-07:
What are you trying to do?

alpceylan, 2020-10-07:
I'm trying to start lime-plotter module which uses network tables

[image]

alpceylan, 2020-10-07:
I'm trying to start lime-plotter module which uses network tables in my localhost

[image]

virtuald, 2020-10-08:
What NetworkTables server are you running on your localhost? Robot, custom thing? What address does it say that it's listening on?

You might try connecting to 'localhost' instead of 127.0.0.1. You can also try your external address?

alpceylan, 2020-10-08:
It's python network tables server. I specify which ip address it should work on, however since i'm working on my personal computer i can only connect to 127.0.0.1.

If I type localhost nothing changes it still try to connect 127.0.0.1

virtuald, 2020-10-08:
Are you sure the server is actually running? Maybe your firewall is blocking it? Are there any errors in the logging output for the server?

alpceylan, 2020-10-08:
I was able to use module before so I'm sure it's running. And I'm sure firewall isn't blocking I tried to run a node.js project in 1735 port and it worked just fine.

I don't know how to check logging output, can you tell me?

virtuald, 2020-10-08:
See the information in the troubleshooting guide: https://robotpy.readthedocs.io/en/stable/troubleshooting.html#problem-i-can-t-determine-if-networktables-has-connected.

Have you tried connecting to OutlineViewer in server mode?

alpceylan, 2020-10-09:
Let me try thanks

virtuald, 2020-10-17:
Did you figure this out?

alpceylan, 2020-10-17:
I'm still trying, I will close this issue when I solved my problem.

alpceylan, 2020-11-16:
error fixed

closed 2020-11-16 by alpceylan

virtuald, 2020-11-16:
What ended up being the problem? We can update the troubleshooting docs to help out someone who has this problem in the future.

alpceylan, 2020-11-16:
it is about java version

alpceylan, 2020-11-16:
it is because of my computer's problems not about the package