robotpy/pynetworktables #121: Is there a way to start the robot in Teleoperated, Autonomous, Practice, or Test mode?

Mockapapella, 2021-09-14:
I'm trying to start my robot in test mode using pynetworktables on Linux (Ubuntu 20.04) for the purposes of building my own custom dashboard. I've initialized a network table server on my RoboRIO with a basic counter that updates every cycle:

```
import wpilib
import ctre
from networktables import NetworkTables
import logging

logging.basicConfig(level=logging.DEBUG)

class MyRobot(wpilib.TimedRobot):
    # Setup networktables
    NetworkTables.initialize()
    smartdash = NetworkTables.getTable("SmartDashboard")
    count = 0
    smartdash.putNumber("Count", count)

    #: update every 0.005 seconds/5 milliseconds (200Hz)
    kUpdatePeriod = 0.005

    def testInit(self):
        """Robot initialization function"""

        self.motor = ctre.WPI_TalonSRX(0)  # initialize the motor as a Talon on channel 0
        self.joyStick = wpilib.Joystick(0)  # initialize the joystick on port 0

    def testPeriodic(self):
        """Runs the motor from a joystick."""

        # Set the motor's output.
        self.motor.set(((self.joyStick.getThrottle() + 1) / 2) * 0.25)
        self.smartdash.putNumber("Count", self.count)
        self.count += 1

if __name__ == "__main__":
    wpilib.run(MyRobot)
```

And then I have a `client.py` file that reads the `Count` table:
```
import sys
import time
from networktables import NetworkTables
import threading
import logging

logging.basicConfig(filename="logs.txt", filemode="a", level=logging.DEBUG)

if len(sys.argv) != 2:
    print("Error: specify an IP to connect to!")
    exit(0)

ip = sys.argv[1]

cond = threading.Condition()
notified = [False]

def connectionListener(connected, info):
    print(info, "; Connected=%s" % connected)
    with cond:
        notified[0] = True
        cond.notify()

NetworkTables.initialize(server=ip)
NetworkTables.addConnectionListener(connectionListener, immediateNotify=True)

with cond:
    print("Waiting")
    if not notified[0]:
        cond.wait()
[... 196 lines cut ...]
DEBUG:root:valueChanged: key: /LiveWindow/Ungrouped/Talon SRX [0]/.controllable; value: False; isNew: 1
DEBUG:nt:notifying '/LiveWindow/Ungrouped/Talon SRX [0]/.controllable' (local=4), flags=5
DEBUG:root:valueChanged: key: /FMSInfo/EventName; value: ; isNew: 1
DEBUG:nt:notifying '/FMSInfo/EventName' (local=7), flags=5
DEBUG:root:valueChanged: key: /FMSInfo/GameSpecificMessage; value: ; isNew: 1
DEBUG:nt:notifying '/FMSInfo/GameSpecificMessage' (local=8), flags=5
DEBUG:root:valueChanged: key: /LiveWindow/.status/LW Enabled; value: True; isNew: 1
DEBUG:nt:notifying '/LiveWindow/.status/LW Enabled' (local=10), flags=5
DEBUG:root:valueChanged: key: /SmartDashboard/Count; value: 339.0; isNew: 1
DEBUG:nt:notifying '/SmartDashboard/Count' (local=9), flags=5
DEBUG:root:valueChanged: key: /FMSInfo/StationNumber; value: 1.0; isNew: 1
DEBUG:nt:notifying '/FMSInfo/StationNumber' (local=12), flags=5
DEBUG:root:valueChanged: key: /LiveWindow/Ungrouped/Talon SRX [0]/Value; value: 0.125; isNew: 1
DEBUG:nt:notifying '/LiveWindow/Ungrouped/Talon SRX [0]/Value' (local=13), flags=5
DEBUG:root:valueChanged: key: /LiveWindow/Ungrouped/.type; value: LW Subsystem; isNew: 1
DEBUG:nt:notifying '/LiveWindow/Ungrouped/.type' (local=14), flags=5
DEBUG:root:valueChanged: key: /FMSInfo/MatchType; value: 1.0; isNew: 1
DEBUG:nt:notifying '/FMSInfo/MatchType' (local=15), flags=5
DEBUG:root:valueChanged: key: /LiveWindow/Ungrouped/Value; value: 0.125; isNew: 1
DEBUG:nt:notifying '/LiveWindow/Ungrouped/Value' (local=16), flags=16
DEBUG:nt:dispatch running 1 connections
DEBUG:nt:dispatch running 1 connections
```

At this point I'm not sure if it's just something small that I've overlooked like a single table value or if it's something else entirely, but I feel like I'm on the right path. Any help would be much appreciated.

virtuald, 2021-09-14:
If I understand your question correctly, you cannot control the robot's mode via NetworkTables. QDriverStation controls the mode of the robot using the DS protocol, not through NetworkTables. You would need to write your own DS code to do this.

Mockapapella, 2021-09-14:
Well that kind of sucks. Just to clarify a few things:

1. The driver station protocol isn't open source, right? So that implementation (and some others I've seen) are all reverse engineered?
2. If I were to participate in a match with my robot, I would be required per FIRST rules to use the official drivers station, but I could use my own custom dashboard displaying networktable data (and other things like camera feeds), right?

TheTripleV, 2021-09-14:
Yes, and yes.

Mockapapella, 2021-09-14:
Alright, thank you for the extra context and the help.

closed 2021-09-14 by Mockapapella